openhwfoundation/programs #559: OBI protocol documentation not found

pcotret, 2022-11-24:
# Template for Documentation Issues
At the OpenHW Group, we know that documentation can have just as many bugs (issues) as code.  So we treat documentation issues
in the same way we treat bugs in code and track them both with GitHub issues.

Please use this file as a template for filing issues against any of the documentation in this repository.
**Hint**: click on the edit symbol (looks like a pencil) and then copy-and-paste the markdown source into your issue.

## Bug Title
OBI protocol documentation may have disappeared.

### Type
Indicate whether the type of problem you found:
* I was looking at the CV32E40X documentation especially for the OBI protocol specs.
> OpenHW Open Bus Interface (OBI) protocol, version 1.5.0, https://github.com/openhwgroup/core-v-docs/blob/master/cores/obi/OBI-v1.5.0.pdf

I understand that `core-v-docs` may have been rename as this `programs` repository. However, I couldn't find the OBI protocol documentation (even if it's based on RST, I could regenerate the PDF by myself).

### Location of Issue
Please provide:
https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/intro.html

### Additional context
We basically have the same issue with other cores, sometimes with older version of OBI specs.
For instance, the CV32E40P : https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/intro.html

pcotret, 2022-11-24:
EDIT: I found the documentation I was looking for in an older commit (https://github.com/openhwgroup/programs/tree/a8e6076c6cfecd57e584f66ce6414cc3b75bf0ca/cores/obi).
However, the issue is still here.

MikeOpenHWGroup, 2022-11-24:
Whoops!   Thanks for reporting this @pcotret.  It seems the OBI spec did not survive the transition from the "core-v-docs" repo to "programs".  I will work with @DBees to resolve this.  There are two aspects to resolving the issue:
- [ ] Restore the OBI spec.
- [ ] Update all links to the new location.

Silabs-ArjanB, 2022-11-25:
@MikeOpenHWGroup @DBees I would like to add that moving OBI documentation under 'programs' is not very logical. Nobody will search it there. Can we keep a core-v-docs as well and keep 'programs' only for 'programs' related stuff?

abukharmeh, 2023-03-07:
Saw @davideschiavone  opened this repo https://github.com/openhwgroup/obi !

pcotret, 2023-03-07:
Great, thanks for the notice @abukharmeh :wink:

MikeOpenHWGroup, 2023-03-07:
Can this issue be closed now?

pcotret, 2023-03-07:
Link has to be updated in each CPU documentation (for instance, https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/intro.html). But I think it's correct for this `programs` repository.

closed 2023-03-07 by pcotret