openhwfoundation/programs #578: obi protocol's

spzeno, 2023-01-29:
# need clarification of obi protocol's *be* and *addr* signals' combination

### Background information

I am writing a obi to axi adapter.

According to obi v1.5 spec, to access one byte at 0x00006001, both below two *be* and *addr* signals' combinations are legal:

1. *be*==4'b0010 and *addr*=32'h0x00006000
2. *be*==4'b0010 and *addr*=32'h0x00006001

(or do I have any misunderstanding?)

However, as I look for PULP's [[per2axi](https://github.com/pulp-platform/per2axi)](https://github.com/pulp-platform/per2axi) module and **[[core2axi](https://github.com/pulp-platform/core2axi)](https://github.com/pulp-platform/core2axi)** module, it seems like they use obi's addr as axi's  addr directly. Thus I think there may need some clarifications.

### Document Template

Choose one of the following as appropriate:

* [[Verficiation Plan template](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/CORE-V_VerifPlan_Template.xlsx)](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/CORE-V_VerifPlan_Template.xlsx)

The above web address is 404 now.

### Document Location

https://github.com/openhwgroup/programs/tree/master/TGs/cores-task-group/obi

MikeOpenHWGroup, 2023-01-29:
Hello @spzeno thanks for the question.  To the best of my knowledge, the PULP team has not adopted the OBI protocol.  This protocol is supported by the OpenHW CV32E40P, CV32E40S and CV32E40X cores and soon the CVE2 will be updated to adopt OBI.  There is no expectation that the PULP team will use it and OpenHW certainly does not expect PULP to use OBI (even though we think it is a great spec!  :wink:).

We are happy to address any OBI questions you may have, but we cannot comment on non-OpenHW use of OBI.

spzeno, 2023-01-30:
I understand. Thanks for your reply.

closed 2023-01-30 by spzeno