openhwfoundation/programs #645: Difficult Navigation

silabs-robin, 2024-01-08:
### Type
Indicate whether the type of problem you found:
* Confusing or poorly worded information.

### Location of Issue
https://github.com/openhwgroup/programs

### Additional context
I was looking for the 40p signoff checklist, and the signoff checklist template, etc, but it was very difficult to locate (even when I knew approximately what I was looking for).
In general, I don't understand this repo, neither the top-level directories nor any sub-level.
The result (I think) is that this repo often gets ignored or avoided.
I am posting this not to vent or to complain, but because I think it could be useful for the openhwgroup to be aware of the problem (and if deemed beneficial, also change it).

See also:
* https://github.com/openhwgroup/programs/issues/549
* https://en.wikipedia.org/wiki/Information_architecture

DBees, 2024-01-10:
That's ok for me @MikeOpenHWGroup @silabs-robin . I can see why it is felt to be buried. I'll move the data via pull request.

MikeOpenHWGroup, 2024-01-10:
Thanks @DBees.  Let's leave this Issue open until your PR happens.

silabs-robin, 2024-01-15:
I can show you have I read the top-level:

* Project-Descriptions-and-Plans
    * "Some administrative openhw stuff, not relevant for me"
* TGs
    * "Some administrative openhw stuff, not relevant for me"
* TWG-and-TG-Attendance-Tracking
    * "Meeting attendance tracking, not relevant for me"
* TWG
    * "Some administrative openhw stuff, not relevant for me"
* dashboard
    * "Maybe we have some CI/CD dashboard and this is a user guide, or maybe its an overview of openhw project status"
* docs
    * "Maybe shows newbeginners what openhw is and gives an overview of projects and repos"
* process
    * "Here I might find "sign-off checklists", because that is a standardized process."
* program
    * "Some administrative openhw stuff, not relevant for me"

DBees, 2024-01-15:
@silabs-robin Thanks a lot for the  feedback on the content here.  I'm delighted to have any suggestions you make and consider how to make this repo more helpful to you and other participants.

Your points:

Project-Descriptions-and-Plans
"Some administrative openhw stuff, not relevant for me"
> This is a set of of directories, each of which corresponds to an OpenHW project. For each project, we have the Project Concept (the initial project idea and motivation to take the project on) and subsequent project gates. OpenHW is member-driven, and the members decide which projects to accept. They also assess projects' progress leading up to release. 
As member of OpenHW, I hope this is relevant for you. Can you consider attending the TWG meetings (held on the 4th Tuesday of each month) to participate in this?

TGs
"Some administrative openhw stuff, not relevant for me"
> The intent here is to provide a place for monthly project reports, organized by TG. 

TWG-and-TG-Attendance-Tracking
"Meeting attendance tracking, not relevant for me"

TWG
"Some administrative openhw stuff, not relevant for me"
dashboard
"Maybe we have some CI/CD dashboard and this is a user guide, or maybe its an overview of openhw project status"
docs
> An OpenHW project dashboard. We could call this "OpenHW-Project-Dashboard" if that is more meaningful.

"Maybe shows newbeginners what openhw is and gives an overview of projects and repos"
process
"Here I might find "sign-off checklists", because that is a standardized process."
program
"Some administrative openhw stuff, not relevant for me"

Can we sync up with a meeting over zoom?
Duncan